bitbank2/PNGdec #1: Strange result when decode 8-bit gray+alpha

spectrenoir06, 2021-08-30:
Hello I can't decode this PNG kapa1.0.png: PNG image data, 25 x 28, 8-bit gray+alpha, non-interlaced
https://static-cdn.jtvnw.net/emoticons/v2/25/default/dark/1.0
In PNGDraw(PNGDRAW* pDraw) I use
```img->png->getLineAsRGB565(pDraw, usPixels, PNG_RGB565_LITTLE_ENDIAN, 0x000000);```

https://github.com/spectrenoir06/Twitch_emote_hub75/blob/63f36034d0e02860393104d5d037c6f8720f9df6/src/main.cpp#L139

bitbank2, 2021-08-30:
What output do you get? The code looks ok.

spectrenoir06, 2021-08-30:
[image]

bitbank2, 2021-09-07:
[image]
I fixed the issue by adding support for gray+alpha to RGB565 in the getLineAsRGB565() function. I'll do a new release of the library too.

closed 2021-09-07 by bitbank2