bitbank2/PNGdec #5: PNG Scaling

mjs513, 2022-01-19:
First want to say been looking for this for a long time and want to say thank you for putting this together.  Used it as soon as I found it!

Now for the question at hand.  Most of my PNGs are larger than what will it in a display and was looking for some sort of scaling ability.  Do you have any plans for a scaling option like you have in the JPEG library.  Or point me to some sort of way to do PNG scaling without having to read the whole image in first?

Thanks 
Mike

bitbank2, 2022-01-19:
Hi Mike,
I'm glad you find the library useful. The reason JPEG has scaling options is because it's a 'natural' consequence of the 8x8 DCT that you can also calculate 4x4, 2x2 and 1x1 from the same coefficients. PNG has no such feature and you'll need to capture + buffer at least a line at a time to do your own scaling. To save memory, put the scaling code in the PNGDRAW callback function. You may also have to increase the PNG_MAX_BUFFERED_PIXELS size in PNGdec.h since images larger than that size will be rejected if you don't.

mjs513, 2022-01-19:
Thanks for the quick reply :)

Was kind of afraid you were going to say that.  Already started reading up on scaling again so at some point will probably update to include it.  Saw the PNG_MAX_BUFFERED_PIXELS.  Right now I am using a Teensy 4.1 from PJRC with a ILI9341 display using an ILI9341 that does clipping so I can display part of an image that is larger so can at least get something showing.

Again thanks for the great library and the fast response.  Now to go check out animated gifs

closed 2022-01-19 by mjs513