bitbank2/PNGdec #6: No  background mix getLineAsRGB565 when PNG_PIXEL_INDEXED and PNG_PIXEL_GRAYSCALE

spectrenoir06, 2022-01-30:
The lib read correctly the tRNS block and store the alpha pPalette[768]
But doesn't use it in getLineAsRGB565 ( only PNG_PIXEL_TRUECOLOR_ALPHA work )

spectrenoir06, 2022-01-30:
I made a dirty fix for my specific use case
https://github.com/spectrenoir06/PNGdec

spectrenoir06, 2022-01-30:
example: https://static-cdn.jtvnw.net/emoticons/v2/1/default/dark/2.0

bitbank2, 2022-01-31:
working on a fix for 8-bit

bitbank2, 2022-01-31:
I added a fix for 8-bit palette images. Please give it a try:

https://github.com/bitbank2/PNGdec/commit/8abf6be9f75c42e6ad1679309812dd48b188c039

bitbank2, 2025-03-02:
I added a full implementation for gray+alpha, 8-bit indexed+alpha and 32-bit. All check if you want to mix a background color or not. I added functional tests to ensure that the output is correct.

closed 2025-03-02 by bitbank2