bitbank2/PNGdec #8: Cannot parse some png fetched from web

sonictl, 2022-05-24:
Thanks for this wel-coded library! 
However, there's a certain type of png files that I didn't manage to decode properly so far.

I received `rc code == 6` when running "rc = png.open()" function in the example of [LittleFS_PNG](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/PNG%20Images/LittleFS_PNG) given in this repo.

In developer tool of Chrome web browser, if you look into the data obtained after GET, some differences can be found.
=== For your re-building this problem ===
The parse_OK png data can GET from [this url](http://api.map.baidu.com/staticimage?width=320&height=240&center=116.403874,39.914888&zoom=16&markers=116.403874,39.914888&markerStyles=l,)
The parse_fail png data can GET from [this url](http://restapi.amap.com/v3/staticmap?location=-60.403874,49.914888&zoom=4&traffic=1&size=320*240&markers=mid,,:-60.403874,49.914888&key=ee95e52bf08006f63fd29bcfbcf21df0)

The code I used for testing:
for PNG fetching: [LittleFS_Web_Jpg](https://github.com/Bodmer/TJpg_Decoder/tree/master/examples/LittleFS/LittleFS_Web_Jpg) (modified for png instead of jpg)
for PNG parsing: https://github.com/Bodmer/TFT_eSPI/blob/master/examples/PNG%20Images/LittleFS_PNG/LittleFS_PNG.ino
The png fetched from url above (http://restapi.amap.com/) is neccessary for me.
Any suggestions and fixing method will be very appreciated! Thank you!!

sonictl, 2022-05-24:
If I download the parse_fail into PC and put that png into <project_root>/data, build the littleFS fileSystem and upload it to ESP32. The parse_fail png will be parsed normally. 

I need to download the png from web, store it into littleFS, parse it from littleFS file. Then, the parse_fail png data will not be decoded.

bitbank2, 2022-05-24:
So you've isolated the problem to be with LittleFS? Sorry, but it sounds like the PNG library can decode the data and it's a streaming problem outside of my code. Can you explain why you think this is a problem with the PNG decoder?

bitbank2, 2022-05-24:
It may be, but my point is that the PNG decoder has no ownership of file handling or streams. The code has callbacks for user code to provide a source of data. If PNG data is delivered to the decoder with the HTTP header preceding the image data, that issue is outside the scope of the PNG decoder.

closed 2022-05-24 by bitbank2