bitbank2/PNGdec #9: Is it possible to make PNG decoder without callbacks?

jhnlmn, 2022-06-20:
Hi
The most popular graphics library now is LVGL and it also works with callbacks: 
rendering callback is called multiple times with different rendering areas (usually from top to bottom)
and I am supposed to render the next portion of the PNG to that rectangle.
The only way I found to marry LVGL with PNGdec (or pngle) is to create a second thread for PNG decoding.
When LVGL calls my callback on the first thread, I have to block it and awaken the PNG thread, which calls PNG decoder
and start receiving callbacks until the rendering area requested by LVGL is filled. Then I freeze PNG task and awaken the first task,
which calls rendering callback again with the next rendering area, etc.
It works, but requires numerous task switches to finish screen rendering - very clumsy.

So, the ideal PNG decoder for low-memory systems will not have any callbacks, but will have API like that:
- open
- get_next_row(char * buffer, int size) 
  or get_next_pixels(char * buffer, int size)
- close
  
Do you think it is possible to write such decoder?
Thank you

bitbank2, 2022-06-20:
What you're asking for is certainly possible, but the code will be a bit more complicated to implement. The reason I did it that way is because it's easier to work from the full file decoding and emit lines as they're ready than to request a line and rebalance the source data stream / tags / flate data. It's a matter of holding on to the file state in a way that allows re-entry into the main decode loop. You can try to rearrange the code I've written to manage that state. Yours is the only request to do this, but you've explained it clearly why this is needed to work with LVGL. Working in memory-constrained environments creates many challenges such as this one. FYI - my SLIC codec works the way you want; perhaps if you're in charge of the compressed graphics you can have a look and see if it fits better.

closed 2022-06-20 by bitbank2

Saddamisalami, 2023-10-14:
Could you create  or point me to an example of this way?

bitbank2, 2023-10-16:
Sorry, I cannot (to both).