bitbank2/PNGdec #11: Alternatives for using a global file variable

lmurdock12, 2022-07-24:
Hi,

I have successfully gotten your library working on an ESP32 that displays to an RGB matrix using @mrfaptastic's Matrix library.

I am trying to find an alternative to having to define my File variable globally such as in: https://github.com/bitbank2/PNGdec/blob/master/examples/sdcard_slideshow/sdcard_slideshow.ino

I am looking to encapsulate the PNG Drawing capability within a higher level image function that I can use for dynamically loading and drawing new images to the LED Matrix. The specific number of class instantiations I will need will never be known at compile time, but I will have multiple files open at once. 

Essentially I want a class like the following:

image1 = ImageManager("filename.png")
image2 = ImageManager("file2.png")

image1.render()
image2.render()

Where ImageManager calls the png.Open() function when initialized. 

I have tried to encapsulate the PNG class and the File variable within a higher level class along with my callback functions inside it but then the compiler complains because the callback functions are not static.
If I set the functions to static I can no longer access the unique File variable that belongs to each instantiation. 

Essentially I want to be able to pass in a File object to the Open() callback without having to explicitly define it as a global variable.

I am not extremely knowledgeable about C++ so maybe I am missing an obvious solution here...

Thanks!

bitbank2, 2022-07-24:
The open callback can allocate a new file handle and return it without using a global variable. Just remember to free it when you close it.

closed 2022-07-24 by bitbank2

cammo1234, 2024-02-10:
Hi @lmurdock12 , sorry to hijack an old issue.

I'm also trying to display png files on the HUB75 LED matrices using this library, and mrfaptastic's library. I stumbled across the two issues you have raised in this repo, and thought I would reach out. I hope that is ok.

I am able to get gifs to play, but I am bashing my head against a wall with pngs. I have written my own callback functions, but wasn't having any luck, so I then copied yours across from the other issue, but still nothing but a blank LED matrix.

Were you ever able to get this project running? Would you be willing to share your code?

Thank you for any help
Cam