bitbank2/PNGdec #13: Encoding png

leetut, 2022-08-21:
How do I encode a png?
Every online byte array tool I've tried prints nothing to the display

bitbank2, 2022-08-21:
It may not be the byte array tool, but the code you're using. Can you share a minimal example which reproduces the problem?
btw - you should use my image_to_c tool --> https://github.com/bitbank2/image_to_c

leetut, 2022-08-21:
using this sketch, i get the panda on my display, i copy just the bits from online conversion and paste them after this: ``
`#include <pgmspace.h>
const uint8_t panda_png[] PROGMEM = {`
into the panda_png.h file, but i only see a color changing background, no image
`
// This example renders a png file that is stored in a FLASH array
// using the PNGdec library (available via library manager).

// Include the PNG decoder library
#include <PNGdec.h>
#include "panda_png.h" // Image is stored here in an 8 bit array

PNG png; // PNG decoder inatance

#define MAX_IMAGE_WDITH 240 // Adjust for your images

int16_t xpos = 0;
int16_t ypos = 0;

// Include the TFT library https://github.com/Bodmer/TFT_eSPI
#include "SPI.h"
#include <TFT_eSPI.h>              // Hardware-specific library
TFT_eSPI tft = TFT_eSPI();         // Invoke custom library

//====================================================================================
//                                    Setup
//====================================================================================
void setup()
{
  Serial.begin(115200);
  Serial.println("\n\n Using the PNGdec library");

  // Initialise the TFT
  tft.begin();
  tft.fillScreen(TFT_BLACK);

  Serial.println("\r\nInitialisation done.");
}

//====================================================================================
//                                    Loop
//====================================================================================
void loop()
{
  int16_t rc = png.openFLASH((uint8_t *)panda_png, sizeof(panda_png), pngDraw);
  if (rc == PNG_SUCCESS) {
    Serial.println("Successfully png file");
    Serial.printf("image specs: (%d x %d), %d bpp, pixel type: %d\n", png.getWidth(), png.getHeight(), png.getBpp(), png.getPixelType());
    tft.startWrite();
    uint32_t dt = millis();
    rc = png.decode(NULL, 0);
    Serial.print(millis() - dt); Serial.println("ms");
    tft.endWrite();
    // png.close(); // not needed for memory->memory decode
  }
  delay(3000);
  tft.fillScreen(random(0x10000));
}

//=========================================v==========================================
//                                      pngDraw
//====================================================================================
// This next function will be called during decoding of the png file to
// render each image line to the TFT.  If you use a different TFT library
// you will need to adapt this function to suit.
// Callback function to draw pixels to the display
void pngDraw(PNGDRAW *pDraw) {
  uint16_t lineBuffer[MAX_IMAGE_WDITH];
  png.getLineAsRGB565(pDraw, lineBuffer, PNG_RGB565_BIG_ENDIAN, 0xffffffff);
  tft.pushImage(xpos, ypos + pDraw->y, pDraw->iWidth, 1, lineBuffer);
}`

bitbank2, 2022-08-21:
What is the serial print output? Is png.openFLASH succeeding? Tell me what the return code is.
Perhaps you are trying to decode an interlaced PNG?

leetut, 2022-08-21:
using this tool:
https://lvgl.io/tools/imageconverter
after copying the bits minus the headers into panda.h and uploading to teensy, serial says: Initialisation done.
nothing else, 

if i copy the bits from your m5stickc_plus_test .h file, into the panda.h file
i get:
Initialisation done.
Successfully png file
image specs: (240 x 135), 8 bpp, pixel type: 2
22ms
and it displays that fine,

think i figured out the image_to_c terminal command
Lees-iMac:MacOS lee$ ./image_to_c 1.png > test.h
but im on MacOS 10.13 and get this error:
dyld: lazy symbol binding failed: Symbol not found: ____chkstk_darwin
  Referenced from: /Users/lee/Desktop/image_to_c-master/dist/MacOS/./image_to_c (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: ____chkstk_darwin
  Referenced from: /Users/lee/Desktop/image_to_c-master/dist/MacOS/./image_to_c (which was built for Mac OS X 10.15)
will try on old windows machine

bitbank2, 2022-08-21:
just rebuild it on linux/windows/macos from the command line (makefile)

closed 2022-08-21 by bitbank2

leetut, 2022-08-21:
Works great after running Makefile, thank you!