bitbank2/PNGdec #15: Error decoding a valid PNG image

matt-42, 2022-11-25:
Hi  @bitbank2,

Used PNGdec with success until I tried it on image created by my Iphone 7: It fails to decode this PNG created on IOS 15.7.1 (and all the other png I generated with this phone):
[image]

Upgrading my iphone and PNGdec did not help. Also tried to update zlib, did not help either.

The code is failing here:
https://github.com/bitbank2/PNGdec/blob/faf03e7597aff70fdd09e1b66c0af306adb9634a/src/inflate.c#L899
```
 if ((hold & 0xffff) != ((hold >> 16) ^ 0xffff)) {
                strm->msg = (char *)"invalid stored block lengths";
                state->mode = BAD;
                break;
            }
```

the same image in base64 just in case: 
iVBORw0KGgoAAAANSUhEUgAAAFAAAABQCAYAAACOEfKtAAAAAXNSR0IArs4c6QAAAERlWElmTU0AKgAAAAgAAYdpAAQAAAABAAAAGgAAAAAAA6ABAAMAAAABAAEAAKACAAQAAAABAAAAUKADAAQAAAABAAAAUAAAAAAx4ExPAAAAHGlET1QAAAACAAAAAAAAACgAAAAoAAAAKAAAACgAAAG1mTb4TQAAAYFJREFUeAHslsFOwkAQhpcrL6SRqyKvITfw4CP4QGo0PIrxzhOQkJBgQjS123TJzLgN7Uwvnf6HTZcys2E+/v9vQwjhGUvN4KlkFwosNYMtANoEBIBG9wEgANosaM1/KBAKhALVrxBW+/XRDwvDwrAwLNxHlgz1DGQgMhAZiAwcan718buRgchAZOBwM/BxHWBhi4WLHwBUP0heX0IBgIb8LOEBoNa+X58VvPEA/D2F4u62v4dVrb5xACTDxoGrpVVe7NvvzurzDzABa7pGZXaFSc8qe/2+xiwfmFLO6qMA6P797TLM44Gf6RoghdN1f32Vh0nPqZXrV4F0WMs+Wfz0/U990f7jATi74QC6QI0QaX2C6hagzKr5nFty88GBUDht9u4BSghk4Gg7tmTtpc+i36eFJQQxNAOYvptM2qky1ddXABRAKriLew4z1sQ/JVPrD6BU33SaHTwHQ3PPP8CMajSgmnoA0AjYF8D1imdXQ241qUlz3xdAmX9GdbUBuv0DAAD//1QPYhQAAAIfSURBVO2YQU7DMBBFo1Zd9EDQE/QUcAVYwQaoEBwAsYaDIE6TSyCgtBDiViPGzo9jO55AsBeR7Unt8X/+k0QtiqKo/stVbYuKXwPoKuscMgC5EN5fXcrkUzp4HtWX0sbWlQFoCrGNPz/iCH170QEuFnHWZbDQgcQHuF3rQmzw2u49PviLN9fqEI5ghMTiAzSFxBqfn9mhmnlGCdAUITlWTueQzFz8nmA/ngOfn/qXrgkhdDyf63BHATBUrMQ8QWCa6+s8cRxog3C9Gt6ZowJ4f2cHhMTEeFPbDg3lFIr1d2AMIdOp/RBsOcx7QqDM0qVxP4Dm5vm4r5DbmzCoffN6zv+7AJEQfkBtfTRPMBYOsE2AigtuWFtblb7KRXsZKi/LEwaQNozayWQ4gEyIBnbAeHyAA27+t6DxvP4Akesolhg8BdIPIIFC7XKZVOmSC+MBTNB9fg5ErqNYovDcAX5tfj4VCBq1pydJlq5fCRMs1CbsPjcHImgUSxxeN8DNe3vpKogZYMdnDDkNtRnezkBlzQE7CUGjWNucBOMY4Po1l66jGTBAchpqHRdO5fnYBIigUSzDazzudIDHR7l0PU2iAySnodZz4fRKGEGjWGJ/kvoc/t6Bhwe5dAMrbA+QnIbawIV9TnHMvy0LBI1is1njrTNmsRJ7twPM7us0UDvADK8TnnI0Bnh14TRZoiTGtiYGmN3nbKAmwAzPGd6uhL8B7+wBoFN36H8AAAAASUVORK5CYII=

bitbank2, 2022-11-26:
iOS PNGs use a different initialization parameter. I didn't add support for them to this library (yet), but have supported them in the past. Search png.inl for "PIL_COMP_IPHONE_FLATE" and you'll see it. If you want to support this option, make the change to the code and submit a PR.

matt-42, 2022-11-29:
Hi @bitbank2,
It was not an ios specific problem but some bugs that prevent PNGdec to read png that have multiple IDAT chunks. I fixed it and will send a pull request later when I'll have time.

bitbank2, 2022-11-29:
oh - thanks for that. I look forward to seeing your fix.

urjaman, 2023-05-20:
> oh - thanks for that. I look forward to seeing your fix.
Still looking forward I suppose, but in case you're curious, i think this is
https://github.com/bitbank2/PNGdec/blob/master/src/png.inl#L768
which feeds the content following the IDAT chunk (CRC and length of next one etc) into inflate, instead of capping it to the current iLen somehow.

Tenkir, 2023-07-25:
I'm encountering this same issue with PNGs from photoshop. Attached is the png I'm attempting to load with the SD slideshow example code.

[image]

matt-42, 2023-07-25:
This fixes decoding of PNGs created on IOS:
https://github.com/bitbank2/PNGdec/pull/19
Hope this fix your issue too.

Tenkir, 2023-07-25:
> This fixes decoding of PNGs created on IOS: #19 Hope this fix your issue too.

Yes this also solves the issue for photoshop PNGs! Thank you!

closed 2024-03-25 by bitbank2