bitbank2/PNGdec #21: Pngdec PNG load works from Flash, not from SD

madaeon01, 2024-01-13:
hi,
i want show in my display the png image stored in sd card, i used PNGDEC and the examples,
it works loading from Flash, LittleFS, but not from SD. It seems the PNG is loaded but no matter what I change, I always get this strange image on the Lcd.
This is the code, modified from example "LittleFS_PNG".

[image]

`// This example if for processors with LittleFS capability (e.g. RP2040,
// ESP32, ESP8266). It renders a png file that is stored in LittleFS
// using the PNGdec library (available via library manager).

// The test image is in the sketch "data" folder (press Ctrl+K to see it).
// You must upload the image to LittleFS using the Arduino IDE Tools Data
// Upload menu option (you may need to install extra tools for that).

// Don't forget to use the Arduino IDE Tools menu to allocate a LittleFS
// memory partition before uploading the sketch and data!

#include <LittleFS.h>
#define FileSys LittleFS

//#include <FS.h>
#include <SD.h>

// Include the PNG decoder library
#include <PNGdec.h>

PNG png;
#define MAX_IMAGE_WIDTH 120// Adjust for your images

int16_t xpos = 0;
int16_t ypos = 0;

// Include the TFT library https://github.com/Bodmer/TFT_eSPI
#include "SPI.h"
#include <TFT_eSPI.h>              // Hardware-specific library
TFT_eSPI tft = TFT_eSPI();         // Invoke custom library

//====================================================================================
//                                    Setup
//====================================================================================
void setup()
{
  Serial.begin(115200);
  Serial.println("\n\n Using the PNGdec library");

/*  // Initialise FS
  if (!FileSys.begin()) {
    Serial.println("LittleFS initialisation failed!");
    wh*ile (1) yield(); // Stay here twiddling thumbs waiting
  }*/

  // Initialise the TFT
  tft.begin();
  tft.fillScreen(TFT_BLACK);

  Serial.println("\r\nInitialisation done.");

    while (!SD.begin(5)) {
    Serial.println("Unable to access SD Card");
    tft.println("Unable to access SD Card");
    delay(1000);
  }

  
}

File pngfile;

void * pngOpen(const char *filename, int32_t *size) {
  Serial.printf("Attempting to open %s\n", filename);
  pngfile = SD.open(filename);
  *size = pngfile.size();
  return &pngfile;
///  Serial.printf("Attempting to open %s\n", filename);
///  pngfile = FileSys.open(filename, "r");
///  *size = pngfile.size();
///  return &pngfile;
}

void pngClose(void *handle) {
  File pngfile = *((File*)handle);
  if (pngfile) pngfile.close();
}

int32_t pngRead(PNGFILE *page, uint8_t *buffer, int32_t length) {
  if (!pngfile) return 0;
  page = page; // Avoid warning
  return pngfile.read(buffer, length);
}

int32_t pngSeek(PNGFILE *page, int32_t position) {
  if (!pngfile) return 0;
  page = page; // Avoid warning
  return pngfile.seek(position);
}
//=========================================v==========================================
//                                      pngDraw
//====================================================================================
// This next function will be called during decoding of the png file to
// render each image line to the TFT.  If you use a different TFT library
// you will need to adapt this function to suit.
// Callback function to draw pixels to the display
void pngDraw(PNGDRAW *pDraw) {
  uint16_t lineBuffer[MAX_IMAGE_WIDTH];
  png.getLineAsRGB565(pDraw, lineBuffer, PNG_RGB565_BIG_ENDIAN, 0xffffffff);
  tft.pushImage(xpos, ypos + pDraw->y, pDraw->iWidth, 1, lineBuffer);
}

//====================================================================================
//                                    Loop
//====================================================================================
void loop()
{
  // Scan LittleFS and load any *.png files
//  File root = LittleFS.open("/", "r");
//  while (File file = root.openNextFile()) {
 //   String strname = file.name();
    String strname = "/EagleEye.png"; ///+ strname;
//    Serial.println(file.name());
//    // If it is not a directory and filename ends in .png then load it
//    if (!file.isDirectory() && strname.endsWith(".png")) {
 //     // Pass support callback function names to library

 
      int16_t rc = png.open(strname.c_str(), pngOpen, pngClose, pngRead, pngSeek, pngDraw);
      if (rc == PNG_SUCCESS) {
        tft.startWrite();
        Serial.printf("image specs: (%d x %d), %d bpp, pixel type: %d\n", png.getWidth(), png.getHeight(), png.getBpp(), png.getPixelType());
        uint32_t dt = millis();
        if (png.getWidth() > MAX_IMAGE_WIDTH) {
          Serial.println("Image too wide for allocated line buffer size!");
        }
        else {
          rc = png.decode(NULL, 0);
          png.close();
        }
        tft.endWrite();
        // How long did rendering take...
        Serial.print(millis()-dt); Serial.println("ms");
      }
  //  }
//    delay(3000);
//    tft.fillScreen(random(0x10000));
//  }
}`

The serial monitor says
" Using the PNGdec library

Initialisation done.
Attempting to open /EagleEye.png
image specs: (120 x 120), 8 bpp, pixel type: 6"

I tired several Png files, this one is from an example (littleFS) and if I load the same file from LittlFS it works.

bitbank2, 2024-01-14:
This issue has nothing to do with PNGDEC; as you saw, it works from FLASH and LittleFS. With embedded devices there are tons of dependencies and if one thing fails, the whole system stops working. The decoder needs to be fed valid data and it will work. The file callbacks are user-defined and not part of my library. I'm sorry, but I can't dedicate time to debug your situation.

closed 2024-01-14 by bitbank2

madaeon01, 2024-01-14:
Is paid support available?

bitbank2, 2024-01-14:
Sei italiano? Si, posso fare un consulting se vuole

madaeon01, 2024-01-14:
Quali sono i costi?
È un progetto privato ma vorrei terminarlo velocemente e questo problema mi
blocca dal continuare.
La sd funziona, se uso l'example sdtest della libreria Sd sembra tutto ok.
Ho visto in un forum che altri hanno avuto un problema simile, sembra, ma
non è chiaro se abbiano risolto.
Inglese o italiano è uguale, grazie.

bitbank2, 2024-01-14:
ok, please zip up your project (I see the code above, but it's better to get the whole thing). Send me the image and a description of the connections to the display and SD card. What MCU are you using? Send info to bitbank@pobox.com

I'm not sure the cost yet. I normally charge $125/hr, but I may see the problem in few minutes.

madaeon01, 2024-01-15:
Hi,
thanks, I found the issue,
https://github.com/Bodmer/TFT_eSPI/issues/1508
"This occurs because the SD and TFT are on different pins so a single SPI
hardware port cannot be connected to both. By default most libraries use
the FPSI port. If you add this line to your TFT_eSPI setup file then the
library will use a different SPI port:

#define USE_HSPI_PORT"

I just added USE_HSPI_PORT to TFT_eSPI setup and now it works fine.

Regards,Davide