bitbank2/PNGdec #22: Cropped picture

PE5PVB, 2024-01-23:
Hi there,

I'm using your library for decoding PNG images received by DAB radio. Now I have an issue with some pictures which are showing cropped.

Here is an example:
`uint8_t pngfile[] = {
  0x89, 0x50, 0x4E, 0x47, 0x0D, 0x0A, 0x1A, 0x0A, 0x00, 0x00, 0x00, 0x0D, 0x49, 0x48, 0x44, 0x52, 0x00, 0x00,
  0x01, 0x40, 0x00, 0x00, 0x00, 0xF0, 0x01, 0x00, 0x00, 0x00, 0x00, 0x59, 0x56, 0x80, 0xC6, 0x00, 0x00, 0x02,
  0x8D, 0x49, 0x44, 0x41, 0x54, 0x68, 0xDE, 0xED, 0xD8, 0x3D, 0x6E, 0xDB, 0x30, 0x14, 0x07, 0xF0, 0x27, 0x70,
  0x60, 0x87, 0x42, 0x5C, 0x3B, 0xB0, 0xE2, 0x35, 0x14, 0xD8, 0x30, 0x0B, 0xF4, 0x22, 0x39, 0x82, 0xBA, 0x79,
  0x08, 0x20, 0x15, 0x1D, 0xBC, 0xD5, 0x17, 0x08, 0xD0, 0x83, 0x74, 0x61, 0x90, 0xA1, 0x5B, 0x7D, 0x84, 0xD2,
  0xF0, 0x05, 0x14, 0x64, 0x61, 0x00, 0x56, 0x2C, 0x29, 0x45, 0x9F, 0xAE, 0x22, 0x0E, 0x2D, 0x9A, 0x06, 0xFC,
  0x4F, 0x82, 0xF4, 0x83, 0xC8, 0xF7, 0x44, 0x99, 0x82, 0x01, 0x42, 0x42, 0x42, 0x42, 0x42, 0x42, 0x42, 0xC8,
  0x51, 0xE5, 0x44, 0x68, 0x4E, 0xE4, 0x22, 0xD4, 0x39, 0x93, 0x1E, 0x10, 0xDF, 0x35, 0x90, 0x09, 0x1F, 0x58,
  0x7A, 0x40, 0xF4, 0xD0, 0x40, 0x5E, 0x2C, 0xC0, 0xE8, 0xA7, 0x85, 0xCA, 0xC2, 0xA5, 0xAA, 0x6B, 0xA8, 0x35,
  0xCF, 0x17, 0xFB, 0x63, 0x1A, 0x58, 0x2D, 0xC2, 0xDC, 0xC2, 0xCA, 0x17, 0x1A, 0xCD, 0xB5, 0xD7, 0xD0, 0x5F,
  0x3C, 0xEE, 0x18, 0x39, 0xF8, 0x43, 0x73, 0xE3, 0x55, 0xF5, 0xC1, 0xA3, 0xEA, 0xBA, 0xE1, 0x07, 0x8F, 0x86,
  0x63, 0x07, 0xBF, 0x59, 0x28, 0x96, 0xA0, 0x7B, 0xD6, 0xB7, 0xF6, 0x59, 0xCB, 0x05, 0x48, 0x1E, 0xE1, 0xE2,
  0x32, 0x63, 0x6E, 0x3D, 0xDE, 0x58, 0x58, 0x2E, 0x36, 0x32, 0x24, 0xE4, 0xAF, 0x87, 0x9B, 0x26, 0x1A, 0x98,
  0x69, 0x0F, 0x89, 0x31, 0x6A, 0x70, 0x61, 0x0C, 0xAB, 0x29, 0xCC, 0x67, 0xA0, 0x99, 0x42, 0x33, 0x07, 0x8B,
  0x31, 0x8C, 0x66, 0xA1, 0x18, 0x43, 0x34, 0x0B, 0xE5, 0x18, 0xE2, 0x59, 0x58, 0x8E, 0x21, 0x99, 0x85, 0x6A,
  0x0C, 0x59, 0x80, 0x01, 0xFE, 0xD7, 0x50, 0x0D, 0x60, 0x81, 0x5A, 0x28, 0xA2, 0x21, 0x14, 0xD3, 0x5F, 0x8A,
  0x17, 0x05, 0xFB, 0xF6, 0x8C, 0xA1, 0x4B, 0xE5, 0x0B, 0x8D, 0x37, 0x2C, 0x7C, 0xA1, 0xF0, 0x85, 0xD2, 0x17,
  0x96, 0x4F, 0x40, 0x77, 0xFE, 0xB7, 0x8B, 0xE2, 0x49, 0xA8, 0x03, 0xFC, 0x37, 0x90, 0xF9, 0x42, 0x32, 0x03,
  0xC9, 0x14, 0xE2, 0x19, 0x78, 0xB6, 0x69, 0xA2, 0x19, 0x78, 0xB6, 0x0D, 0x47, 0x33, 0xF0, 0x6C, 0x63, 0x87,
  0x19, 0x78, 0xF6, 0xA9, 0xD0, 0x7D, 0x64, 0x4C, 0x61, 0x3E, 0x85, 0x7C, 0x06, 0xF2, 0x29, 0x6C, 0x67, 0x3E,
  0x85, 0xED, 0x44, 0x5F, 0x24, 0x1C, 0xAF, 0x9E, 0xBE, 0x21, 0x0B, 0xEB, 0x31, 0xC0, 0x00, 0x9F, 0x39, 0x54,
  0x7F, 0x1C, 0x96, 0xBE, 0x50, 0xFA, 0x42, 0xE1, 0x0B, 0x0B, 0x5F, 0xD8, 0xBE, 0xE3, 0xEC, 0x1C, 0xAA, 0xE1,
  0xF9, 0xE5, 0xBF, 0x77, 0x42, 0x42, 0x9E, 0x73, 0x90, 0x4A, 0xD1, 0xCE, 0xD0, 0x48, 0x20, 0x05, 0x32, 0xA6,
  0x09, 0x57, 0x91, 0x46, 0x8A, 0x32, 0x11, 0x69, 0x56, 0x66, 0xE6, 0xD3, 0xBE, 0x7D, 0x17, 0x88, 0x5E, 0xE3,
  0x1A, 0x62, 0xED, 0xE0, 0x9A, 0x2B, 0x54, 0x61, 0x4D, 0x79, 0x89, 0x2A, 0xA6, 0x33, 0x73, 0xD3, 0x41, 0x7B,
  0x0D, 0xEF, 0xC0, 0xC2, 0x98, 0x3A, 0xB8, 0x4D, 0x01, 0xED, 0x49, 0x42, 0xB7, 0x97, 0x38, 0xA6, 0x57, 0xD9,
  0xE9, 0x58, 0x94, 0x2D, 0x84, 0x0D, 0x69, 0x20, 0x38, 0x98, 0xA5, 0x80, 0x71, 0x1C, 0xBF, 0xCE, 0x00, 0x03,
  0x4D, 0x32, 0x79, 0x2C, 0x64, 0x0F, 0xD9, 0x00, 0x4A, 0x0B, 0xD1, 0x67, 0xFC, 0xAA, 0x86, 0x71, 0x26, 0x3F,
  0x74, 0x90, 0x14, 0xBC, 0x81, 0xA4, 0xB0, 0xF0, 0xAD, 0x70, 0x77, 0xBC, 0xC6, 0x6F, 0xAE, 0x00, 0x0B, 0x07,
  0xEF, 0x3A, 0x88, 0xCB, 0xD5, 0x6A, 0x57, 0xB9, 0x62, 0x4A, 0x0B, 0x99, 0xB0, 0x55, 0x1C, 0xAE, 0x31, 0x70,
  0xDB, 0x05, 0x4A, 0xB3, 0x63, 0x3F, 0x34, 0xD2, 0xAB, 0xAF, 0x35, 0x44, 0xAE, 0x6A, 0x26, 0x2D, 0x34, 0x16,
  0x32, 0x19, 0xD5, 0x55, 0xF7, 0x10, 0x36, 0x17, 0xF7, 0xF5, 0xD0, 0xB0, 0xB1, 0x30, 0xB1, 0x73, 0x84, 0x7B,
  0x0B, 0xA3, 0x12, 0x18, 0x53, 0xB6, 0x3D, 0x3D, 0x8C, 0x2F, 0x4E, 0x0D, 0x8C, 0x1F, 0xAB, 0x06, 0x9C, 0x60,
  0x00, 0x5B, 0x0D, 0x2B, 0x32, 0xF3, 0x71, 0x00, 0xD3, 0x21, 0xDC, 0x3A, 0x18, 0xC7, 0x35, 0xA4, 0x90, 0x9D,
  0x7A, 0x18, 0x27, 0xE9, 0x6D, 0xD3, 0x9E, 0x1A, 0xAE, 0x53, 0x88, 0x08, 0x49, 0x20, 0xCB, 0x30, 0xB1, 0xF0,
  0xD8, 0x0F, 0xCD, 0x74, 0xFA, 0x7E, 0x67, 0xD8, 0x83, 0xD9, 0x6B, 0x30, 0xDF, 0x29, 0x6D, 0x1E, 0x21, 0xE4,
  0x92, 0x54, 0x63, 0x88, 0xED, 0xA2, 0xD8, 0x3B, 0xB8, 0x53, 0x0E, 0xC6, 0xCD, 0xA2, 0x70, 0xED, 0xD1, 0x16,
  0xCA, 0x93, 0xE8, 0xE6, 0x28, 0xDA, 0x83, 0xEE, 0xCC, 0x25, 0xBC, 0x83, 0xBA, 0x9C, 0x90, 0x90, 0x90, 0x90,
  0x90, 0x90, 0x90, 0xC7, 0xFC, 0x02, 0xD2, 0x9B, 0x28, 0xFD, 0x2E, 0x8B, 0xF9, 0xFD, 0x00, 0x00, 0x00, 0x00,
  0x49, 0x45, 0x4E, 0x44, 0xAE, 0x42, 0x60, 0x82
};`

bitbank2, 2024-01-23:
Not cropped, squished. The getLineAsRGB565() code was missing the case for 1-bpp grayscale (not indexed). The output of the decoder was correct, it's just the helper function I provided to turn the output into RGB565 that had the problem. I just pushed a fix, please give it a try and let me know.

PE5PVB, 2024-01-24:
Yes, this did the trick. Many thanks.

bitbank2, 2024-01-24:
Thanks for reporting and verifying the fix. I'll do a new release.

closed 2024-01-24 by bitbank2