bitbank2/PNGdec #23: Is there an image width limit of 640 pixels?

rtek1000, 2024-02-25:
**Before you create this issue**
This library takes compressed PNG data and outputs pixels to your callback functions. If your problem is related to LCD displays, SD cards, or any other issue that has nothing to do with this library, please stop here. If you need help with your private or commercial project, please email me (bitbank@pobox.com) to ask about paid consulting. Github issues are for submitting actual bugs.

**Describe the bug**
A clear and concise description of what the bug is.

An image of width 641 pixels is not decoded, but it is decoded if the width is 640.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to: https://notisrac.github.io/FileToCArray/ and convert an 641x800 image

2. Compile de code: https://github.com/Bodmer/TFT_eSPI/blob/master/examples/PNG%20Images/Flash_PNG/Flash_PNG.ino
- - Change the MAX_IMAGE_WIDTH to 900

3. See error
- When the problem occurs (above 640 width) the message "Successfully opened png file" does not appear on the serial monitor.

**Expected behavior**
A clear and concise description of what you expected to happen.

- A 480x800 image is displayed. While an 800x480 image is not. 
- I don't know if it's a problem with the image converter website or this PNG library.
- Do you have another converter to test?

**Image you're having trouble with**
If applicable, add original image files to help explain your problem.

**Additional context**
Add any other context about the problem here.

https://github.com/Bodmer/TFT_eSPI/issues/3185

bitbank2, 2024-02-25:
Yes, this is to reduce RAM needed on MCUs. You can change it - see PNGdec.h:

/* Defines and variables */
#define PNG_FILE_BUF_SIZE 2048
// Number of bytes to reserve for current and previous lines
// Defaults to 640 32-bit pixels max width
#define PNG_MAX_BUFFERED_PIXELS (640*4 + 1)

closed 2024-02-25 by bitbank2

rtek1000, 2024-02-25:
How is PNG_FILE_BUF_SIZE related to PNG_MAX_BUFFERED_PIXELS?

I increased the value of PNG_MAX_BUFFERED_PIXELS, but it decreased the value of PNG_FILE_BUF_SIZE, and even so it is now working with an 800x480 image.

```C++
/* Defines and variables */
#define PNG_FILE_BUF_SIZE 1024
// Number of bytes to reserve for current and previous lines
// Defaults to 640 32-bit pixels max width
#define PNG_MAX_BUFFERED_PIXELS (800*4 + 1)
```

bitbank2, 2024-02-25:
Sorry, the max buffered pixels needs to be large enough to hold 2 lines (current and previous). If you're decoding 32-bit pixels, then that's 800 x 2 x 4

rtek1000, 2024-02-25:
> Sorry, the max buffered pixels needs to be large enough to hold 2 lines (current and previous). If you're decoding 32-bit pixels, then that's 800 x 2 x 4

So if '800 x 2 x 4' = '6400', it should look like this, as shown below?

```C++
#define PNG_FILE_BUF_SIZE 6400 // 800 x 2 x 4
#define PNG_MAX_BUFFERED_PIXELS (800*4 + 1)
```

Or should it look like this, as shown below?

```C++
#define PNG_FILE_BUF_SIZE 2048
#define PNG_MAX_BUFFERED_PIXELS (800*2*4 + 1)
```

----

Does the value of PNG_FILE_BUF_SIZE not matter?

bitbank2, 2024-02-25:
MAX_BUFFERED_PIXELS is for the pixels, FILE_BUF is for uncompressed data. A small file buffer size will slow it down due to more calls to read more data.

bitbank2, 2024-02-25:
There is TRUECOLOR (24-bpp) and TRUECOLOR with ALPHA (32-bpp). You can use my imageinfo tool to show the difference: https://github.com/bitbank2/imageinfo

rtek1000, 2024-02-25:
> There is TRUECOLOR (24-bpp) and TRUECOLOR with ALPHA (32-bpp). You can use my imageinfo tool to show the difference: https://github.com/bitbank2/imageinfo

Thank you,

I found a website: Get All Metadata Info Of Your Files https://www.metadata2go.com/view-metadata

Apparently the image is 8 bits, which is why it is displayed with PNG_MAX_BUFFERED_PIXELS of '800x4+1' instead of '800x2x4+1':

[image]

bitbank2, 2024-02-25:
bit depth = 8 can mean 24 or 32-bit (R=8, G=8, B=8, A=8)
color_type = indexed = palette = 8 bits or less total
your image has color_type RGB with Alpha = 32 bits per pixel

rtek1000, 2024-02-25:
> bit depth = 8 can mean 24 or 32-bit (R=8, G=8, B=8, A=8) color_type = indexed = palette = 8 bits or less total your image has color_type RGB with Alpha = 32 bits per pixel.

Yes, thank you. I found this other [website](https://compress-or-die.com/Understanding-PNG) that explains about this (Interesting how the small images do not show such significant visible differences)

Can I consider the '800x4+1' setting as appropriate since the converter website is generating 8-bit data?

When the "Treat as binary" option is not active and 32 bits is selected, then the data is 8 bits x 4 in length:

- "Treat as binary" data: **0x89**
- The 32 bit RGBA data: **0x6e97d0ff**

[image]

bitbank2, 2024-02-25:
Use my image_to_c tool to convert files for Arduino. No options needed, the data comes out the way it should.