bitbank2/PNGdec #25: [RP2040, ARDUINO, PLATFORMIO] PNGDec decode causes infinite loop

basgoossen, 2024-05-28:
**Describe the bug**
When running pngDEC with the example octocat files on a RP2040 (platformIO, both earlephilhower cora and pico core). the png header is read correctly, however a call to png.decode(nullptr, 0); results in an infinite loop (in zlib inflate). 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a project in platform IO for the rp2040 (pico core or earlephilhower).
2. include PNGdec.h and an octocat.h file of preference.
3. call png.openRAM(octocat, len, drawPNGLine);
4. verify returned value to be 0 (PNG_SUCCESS);
5. run png.decode(nullptr,0);
6. Device will be stuck in an infinite loop from line: err = inflate(&d_stream, Z_NO_FLASH, iOptions & PNG_CHECK_CRC);

**Expected behavior**
The expected behaviour whould be that the png is decoded and for each line of image drawPNGLine is called.

**Image you're having trouble with**
example Octocat files from the PNGdec library.

basgoossen, 2024-05-28:
Correction, it does not enter an infinite loop, it just crashes the RP2040. (infinite loop was created by adding debug lines and the omitted {} in the infinite for loop / switch loop in inflate.

bitbank2, 2024-05-28:
I believe this is due to the RP2040 having an exception for unaligned pointer access. I think I regressed this code to undo that optimization. Clone the latest code instead of using the last release and see if it's resolved. If not, try removing 
"ALLOWS_UNALIGNED" from inffast.c

basgoossen, 2024-05-29:
First of all thanks for your reply!
I've tested both proposed solutions, they however do not change the situation. The master in Github indeed is different from the library version that PlatformIO offers, but also the newer version (directly cloned from github) still causes the freeze in the same location. I'm actually quite confused why, since it seems like a pretty ligit write to a initialized pointer location.

bitbank2, 2024-05-29:
Did you change the current code on Github to get it to work? When you say "removed the unaligned blocks" do you mean you disabled the #define or actually deleted the lines? The code on Github should work unchanged on your RP2040. If you confirm this, I'll do a new release.

basgoossen, 2024-05-29:
Hi BitBank2,
First i removed the entire #ifdef blocks, but later i also tested removing the 2 occasions of:

#define ALLOWS_UNALIGNED

In both inflate.c and inffast.c, and i can confirm that the latter also works.

Thanks again for your efforts and for creating this tool.

MagnusThome, 2024-11-07:
Any time schedule for a release with this fixed? 

This code example that is included with TFT_eSPI uses PNGdec and fails on rp2040:
https://github.com/Bodmer/TFT_eSPI/blob/master/examples/PNG%20Images/Flash_PNG/Flash_PNG.ino

MagnusThome, 2024-11-07:
Downloaded the current code and it worked fine 👍👍👍

bitbank2, 2024-11-07:
Great, thanks for testing. I'll do the release...

closed 2024-11-07 by bitbank2