bitbank2/PNGdec #28: pngDEC 1.0.3. crashes system on rp2040

anttike, 2024-10-15:
System: Custom rp2040 board with ST7789V display and TFT_eSPI library (PIO SPI).

With pngDEC 1.0.0 and 1.0.1 everything works nicely. When using pngDEC 1.0.2 system sometimes jams and sometimes only background colour is visible but not image itself.

To reproduce use TFT_eSPI example PNG images -> Flash_PNG.

anttike, 2024-10-15:
Sorry for being unclear.  I fully understand that this code was not written for TFT_eSPI.  

I do have custom hardware, but basically it is just generic rp2040. I really like your lib and since it already worked well with versions 1.0.0 and 1.0.1, I thought it might be possible to get to work with 1.0.2 also.

Working code example is just TFT-eSPI library's code example "Flash_PNG". That happily works on 1.0.0 and 1.0.1. but not 1.0.2. So far I have not managed to get it working no matter what I change so for 1.0.2. I do not have working code example.

I did try once to remove ALLOWS_UNALIGNED macros. It changed behaviour so that system does not jam anymore, but then images are not visible. For example if define green display background and put png on top of it, I see only green screen.

bitbank2, 2024-10-15:
I just pushed a change which excludes ALLOWS_UNALIGNED for the RP2040. As I suspected, this was causing the exception. There is another issue to resolve with decoding that panda image. It returns a decoding error for me, so I need to trace why. Try other PNG images while I figure out what specifically is causing the error with the panda image.

bitbank2, 2024-10-15:
I tried to debug the problem in Xcode, but it doesn't fail there (panda image). Same with ESP32. It appears to only fail on the RP2040 using the same code. I don't have a good way to debug code on the RP2040 (single step). I can resolve almost all code issues on my M1 Mac in Xcode, except when there's a compiler bug or some weird behavior. Not sure how to proceed with this issue.

closed 2024-10-15 by bitbank2