bitbank2/PNGdec #29: Allow compile without Arduino.h

geiseri, 2024-10-17:
Can you add the same code as [JPEGDEC](https://github.com/bitbank2/JPEGDEC/blob/07031ddf9286d9831265603c351d79b9e5fc8618/src/JPEGDEC.h#L16) so it can compile with bare idf support.  Thanks!

bitbank2, 2024-10-17:
It already does - at the top of PNGdec.h is this line:

#if defined( __MACH__ ) || defined( __LINUX__ ) || defined( __MCUXPRESSO )

Define any of those macros and it will include the std c libraries instead of Arduino.h

closed 2024-10-17 by bitbank2

geiseri, 2024-10-17:
That didn't seem to catch it with the ESP32 IDF.  What does `MACH`  define as?

bitbank2, 2024-10-18:
There are underscores that turn into bold here when I copied the text. You need to define one of those macros in your compiler flags. Mach doesn't define as anything, it's the MacOS compiler define to indicate the type of kernel.

geiseri, 2024-10-18:
Kinda funny to have one of those defines for ESP32, but I will do that.  I want to use both JPEG and PNG so `__MCUXPRESSO` should work.  Maybe rename that to `NO_ARDUINO` for clarity sake.  Thanks for the awesome libraries too!