bitbank2/PNGdec #34: 1.1.1 is breaking on projects that doesn't use bb_lcd_spi

bmorcelli, 2025-04-20:
I am using this library in a platformio project, where libraries must be installed manually.

The helper file is calling an unnecessary Library for the project (bb_lcd_spi), and mostly will break into many other projects pipelines

You can set it as prerequisite in the library files to keep backward compatibility or have the helper completely commented.

Thank you for your attention and congratulations for the great work you are doing

bitbank2, 2025-04-21:
Thanks for letting me know. I have an idea how to fix it. I'll update here when it's ready.

bitbank2, 2025-04-21:
ok, I pushed a fix. Please clone and see if it corrects the issue. If so, I'll do a new release.

bmorcelli, 2025-04-21:
Nice!! 

It worked flawlessly!

Thank you very much!

closed 2025-04-21 by bmorcelli