bitbank2/PNGdec #35: **Some** OSM tiles decoding gives an error code 2

CelliesProjects, 2025-05-01:
@bitbank2 Hello! Running into some decoding issues again. 

OSM tiles decoding gives an error code 2 sometimes on different tiles. Most of the time the tiles decode OK, but sometimes they give an error.

```
[  3108][I][OpenStreetMap-esp32.cpp:211] updateCache(): submitting 9 jobs
[  3115][I][OpenStreetMap-esp32.cpp:523] tileFetcherTask(): core 1 fetching tile z=15 x=16847, y=10896
[  3126][I][OpenStreetMap-esp32.cpp:523] tileFetcherTask(): core 0 fetching tile z=15 x=16848, y=10896
[  3574][E][OpenStreetMap-esp32.cpp:526] tileFetcherTask(): Tile fetch failed: Decoding https://tile.openstreetmap.org/15/16848/10896.png failed with code: 2
[  3588][I][OpenStreetMap-esp32.cpp:523] tileFetcherTask(): core 0 fetching tile z=15 x=16849, y=10896
[  3593][I][OpenStreetMap-esp32.cpp:523] tileFetcherTask(): core 1 fetching tile z=15 x=16847, y=10897
[  4042][I][OpenStreetMap-esp32.cpp:523] tileFetcherTask(): core 1 fetching tile z=15 x=16848, y=10897
[  4072][I][OpenStreetMap-esp32.cpp:523] tileFetcherTask(): core 0 fetching tile z=15 x=16849, y=10897
[  4485][I][OpenStreetMap-esp32.cpp:523] tileFetcherTask(): core 1 fetching tile z=15 x=16847, y=10898
[  4517][I][OpenStreetMap-esp32.cpp:523] tileFetcherTask(): core 0 fetching tile z=15 x=16848, y=10898
[  4923][I][OpenStreetMap-esp32.cpp:523] tileFetcherTask(): core 1 fetching tile z=15 x=16849, y=10898
[  5237][I][OpenStreetMap-esp32.cpp:488] decrementActiveJobs(): jobs done
[  5267][W][OpenStreetMap-esp32.cpp:262] composeMap(): Tile (z=15, x=16848, y=10896) not found in cache
```

**Image you're having trouble with**
[https://tile.openstreetmap.org/15/16848/10896.png](https://tile.openstreetmap.org/15/16848/10896.png)

As said, most of the time they decode ok, just the occasional fail.

**Additional context**
I see this with more tiles, I might give more examples if needed.

I reported some non decoding tiles a while back in this issue:
https://github.com/CelliesProjects/OpenStreetMap-esp32/issues/4

bitbank2, 2025-05-01:
That image decodes fine with the current code. Perhaps there are some external conditions causing the problem.

CelliesProjects, 2025-05-03:
@bitbank2 The issue was indeed caused by some local factors.

closed 2025-05-03 by CelliesProjects